hideakitai/MsgPacketizer #1: #include <MsgPacketizer.h> in multiple source files

dwjbosman, 2020-08-10:
I have 2 cpp files both contain a line #include<MsgPacketizer.h>

During linking I get an error:

> tmp/arduino-sketch-CBB01DD9DE742798D756B54022239E3C/sketch/plg_rtd.cpp.o (symbol from plugin): In function `arx::debug::log_level':
(.text+0x0): multiple definition of `arx::debug::log_level'
/tmp/arduino-sketch-CBB01DD9DE742798D756B54022239E3C/sketch/iMaxi2Sensor.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
/tmp/arduino-sketch-CBB01DD9DE742798D756B54022239E3C/sketch/plg_rtd.cpp.o (symbol from plugin): In function `arx::debug::log_level':
(.text+0x0): multiple definition of `arx::debug::b_only_sd'
/tmp/arduino-sketch-CBB01DD9DE742798D756B54022239E3C/sketch/iMaxi2Sensor.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
/tmp/arduino-sketch-CBB01DD9DE742798D756B54022239E3C/sketch/plg_rtd.cpp.o (symbol from plugin): In function `arx::debug::log_level':
(.text+0x0): multiple definition of `arx::debug::b_auto_save'
/tmp/arduino-sketch-CBB01DD9DE742798D756B54022239E3C/sketch/iMaxi2Sensor.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
/tmp/arduino-sketch-CBB01DD9DE742798D756B54022239E3C/sketch/plg_rtd.cpp.o (symbol from plugin): In function `arx::debug::log_level':
(.text+0x0): multiple definition of `arx::debug::logger'
/tmp/arduino-sketch-CBB01DD9DE742798D756B54022239E3C/sketch/iMaxi2Sensor.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
/tmp/arduino-sketch-CBB01DD9DE742798D756B54022239E3C/sketch/plg_rtd.cpp.o (symbol from plugin): In function `arx::debug::log_level':
(.text+0x0): multiple definition of `arx::debug::stream'
/tmp/arduino-sketch-CBB01DD9DE742798D756B54022239E3C/sketch/iMaxi2Sensor.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

hideakitai, 2020-08-12:
Ahh... ok, I should try to fix that but maybe it takes some time...
Can you show me another multiple definitions without `#define MSGPACKETIZER_ENABLE_DEBUG_LOG` ?

dwjbosman, 2020-08-12:
In the end I hacked it by creating a  define DEBUG_EXTERN and if set replacing all declarations with extern definitions.

Unfortunately It seems I can't use Debugging and MSgPacketizer together because too much RAM is needed on my 2048 byte Arduino :(

hideakitai, 2020-09-02:
@dwjbosman Sorry for the late reply, fix this issue in v0.2.15. Thanks!

closed 2020-09-02 by hideakitai