hideakitai/MsgPacketizer #3: Subscribe to binary data (MsgPack::bin_t<uint8_t>)

stwirth, 2021-01-04:
I'm trying to subscribe to an array of `uint_8t` on an arduino by changing one line in the example:
```C++
#include <MsgPacketizer.h>

// input to msgpack
int i;
float f;
MsgPack::str_t s; // std::string or String
// MsgPack::arr_t<int> v; // std::vector or arx::vector
MsgPack::bin_t<uint8_t> v; // <-- array of bytes
MsgPack::map_t<String, float> m; // std::map or arx::map

uint8_t recv_index = 0x12;
uint8_t send_index = 0x34;

void setup()
{
    Serial.begin(115200);

    // update received data directly
    MsgPacketizer::subscribe(Serial, recv_index, i, f, s, v, m);

    // publish varibales periodically (default 30[times/sec])
    MsgPacketizer::publish(Serial, send_index, i, f, s, v, m);
}

void loop()
{
    // must be called to trigger callback and publish data
    MsgPacketizer::update();
}
```
I get the following error during compilation:
```
Compiling sketch...
/home/stwirth/src/arduino-1.8.13/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\"" -I/home/stwirth/src/arduino-1.8.13/hardware/arduino/avr/cores/arduino -I/home/stwirth/src/arduino-1.8.13/hardware/arduino/avr/variants/leonardo -I/home/stwirth/Arduino/libraries/MsgPacketizer /tmp/arduino_build_12483/sketch/msgpack_test.ino.cpp -o /tmp/arduino_build_12483/sketch/msgpack_test.ino.cpp.o
In file included from /home/stwirth/Arduino/libraries/MsgPacketizer/util/MsgPack/MsgPack.h:11:0,
                 from /home/stwirth/Arduino/libraries/MsgPacketizer/MsgPacketizer.h:13,
                 from /home/stwirth/Arduino/msgpack_test/msgpack_test.ino:1:
/home/stwirth/Arduino/libraries/MsgPacketizer/util/MsgPack/MsgPack/Unpacker.h: In instantiation of 'void ht::serial::msgpack::Unpacker::deserialize(First&, Rest&& ...) [with First = arx::vector<char, 128>; Rest = {arx::map<String, float, 8>&}]':
/home/stwirth/Arduino/libraries/MsgPacketizer/util/MsgPack/MsgPack/Unpacker.h:71:13:   recursively required from 'void ht::serial::msgpack::Unpacker::deserialize(First&, Rest&& ...) [with First = float; Rest = {String&, arx::vector<char, 128>&, arx::map<String, float, 8>&}]'
/home/stwirth/Arduino/libraries/MsgPacketizer/util/MsgPack/MsgPack/Unpacker.h:71:13:   required from 'void ht::serial::msgpack::Unpacker::deserialize(First&, Rest&& ...) [with First = int; Rest = {float&, String&, arx::vector<char, 128>&, arx::map<String, float, 8>&}]'
/home/stwirth/Arduino/libraries/MsgPacketizer/MsgPacketizer.h:469:13:   required from 'ht::serial::msgpacketizer::subscribe(ht::serial::msgpacketizer::StreamType&, uint8_t, Args&& ...)::<lambda(const uint8_t*, uint8_t)> [with Args = {int&, float&, String&, arx::vector<char, 128>&, arx::map<String, float, 8>&}; uint8_t = unsigned char]'
/home/stwirth/Arduino/libraries/MsgPacketizer/MsgPacketizer.h:469:54:   required from 'struct ht::serial::msgpacketizer::subscribe(ht::serial::msgpacketizer::StreamType&, uint8_t, Args&& ...) [with Args = {int&, float&, String&, arx::vector<char, 128>&, arx::map<String, float, 8>&}; ht::serial::msgpacketizer::StreamType = Stream; uint8_t = unsigned char]::<lambda(const uint8_t*, uint8_t)>'
/home/stwirth/Arduino/libraries/MsgPacketizer/MsgPacketizer.h:464:30:   required from 'void ht::serial::msgpacketizer::subscribe(ht::serial::msgpacketizer::StreamType&, uint8_t, Args&& ...) [with Args = {int&, float&, String&, arx::vector<char, 128>&, arx::map<String, float, 8>&}; ht::serial::msgpacketizer::StreamType = Stream; uint8_t = unsigned char]'
/home/stwirth/Arduino/msgpack_test/msgpack_test.ino:18:63:   required from here
/home/stwirth/Arduino/libraries/MsgPacketizer/util/MsgPack/MsgPack/Unpacker.h:70:13: error: no matching function for call to 'ht::serial::msgpack::Unpacker::unpack(arx::vector<char, 128>&)'
             unpack(first);
             ^~~~~~
/home/stwirth/Arduino/libraries/MsgPacketizer/util/MsgPack/MsgPack/Unpacker.h:134:14: note: candidate: template<class T> typename arx::stdx::enable_if<arx::stdx::is_same<T, ht::serial::msgpack::object::nil_t>::value>::type ht::serial::msgpack::Unpacker::unpack(T&)
         auto unpack(T& value)
              ^~~~~~
```
I'm using MsgPacketizer 0.2.20.

What am I doing wrong?

hideakitai, 2021-01-04:
Hi, could you give me more information about your board (Uno?), firmware version, and IDE (Arduino IDE or PlatformIO)?

stwirth, 2021-01-04:
Sure, sorry for not providing everything.
The board is a KeeYees Pro Micro ATmega32U4 5V 16MHz Micro USB, the Arduino IDE detects it as "Arduino Leonardo". I'm using Arduino IDE 1.8.13 on Ubuntu 16.04. What do you mean by firmware version? How can I retrieve that?

hideakitai, 2021-01-04:
Thanks! Firmware version means the version of Leonard firmware. You can see it from the board manager. https://www.arduino.cc/en/guide/cores

stwirth, 2021-01-04:
This?
[image]

hideakitai, 2021-01-04:
@stwirth Yes! And fixed compile error in v0.2.21. Please check if it works.

stwirth, 2021-01-04:
Wow, that was a quick turnaround! Compiles fine now, thank you!

closed 2021-01-04 by stwirth