hideakitai/MsgPacketizer #16: Cannot compile library on Mega 2560

KennethThompson, 2024-02-03:
Attempting to compile any example on a 2560 produces numerous compile errors. 

Try the following code with 2560 set as a target:

```

#include <ArduinoJson.h>  // include before MsgPacketizer.h

#define MSGPACKETIZER_DEBUGLOG_ENABLE
#include <MsgPacketizer.h>
#include <Ethernet.h>

const uint8_t msg_index = 0x12;

// Ethernet stuff
uint8_t mac[] = {0xAB, 0xCD, 0xEF, 0x01, 0x23, 0x45};
const IPAddress ip(192, 168, 0, 201);
// Ethernet with useful options
// const IPAddress dns (192, 168, 0, 1);
// const IPAddress gateway (192, 168, 0, 1);
// const IPAddress subnet (255, 255, 255, 0);

const char* host = "192.168.0.201";  // loop back
const int port = 54321;

EthernetUDP client;

void setup() {
    Serial.begin(115200);
    delay(2000);

    // Ethernet stuff
    Ethernet.begin(mac, ip);
    // Ethernet with useful options
    // Ethernet.begin(mac, ip, dns, gateway, subnet); // full
    // Ethernet.setRetransmissionCount(4); // default: 8[times]
    // Ethernet.setRetransmissionTimeout(50); // default: 200[ms]

    // start client
    client.begin(port);

    // only subscribing ArduinoJson with lambda is allowed
    // because subscribing JsonDocument as global object is not reccomended
    // see https://arduinojson.org/v6/how-to/reuse-a-json-document/
    MsgPacketizer::subscribe(client, msg_index,
        [&](const StaticJsonDocument<200>& doc) {
            Serial.print(doc["us"].as<uint32_t>());
            Serial.print(" ");
            Serial.print(doc["usstr"].as<String>());
            Serial.print(" [");
            Serial.print(doc["now"][0].as<double>());
            Serial.print(" ");
            Serial.print(doc["now"][1].as<double>());
            Serial.println("]");
        });

    // you can also use DynamicJsonDocument if you want
    // MsgPacketizer::subscribe(client, msg_index,
    //     [&](const DynamicJsonDocument& doc) {
    //         Serial.print(doc["us"].as<uint32_t>());
    //         Serial.print(" ");
    //         Serial.print(doc["usstr"].as<String>());
    //         Serial.print(" [");
    //         Serial.print(doc["now"][0].as<double>());
    //         Serial.print(" ");
    //         Serial.print(doc["now"][1].as<double>());
    //         Serial.println("]");
    //     });

    // or of course you can bind to variables directly
[... 126 lines cut ...]
                                        ~~~~~~~~~~~~~^~~~~~
In file included from /home/ken/Arduino/libraries/DebugLog/DebugLog/util/ArxTypeTraits/ArxTypeTraits.h:36:0,
                 from /home/ken/Arduino/libraries/DebugLog/DebugLog.h:14,
                 from /home/ken/Arduino/libraries/MsgPacketizer/MsgPacketizer.h:64,
                 from /tmp/.arduinoIDE-unsaved202413-2067-80ztgm.3pekm/sketch_feb3a/sketch_feb3a.ino:4:
/home/ken/Arduino/libraries/DebugLog/DebugLog/util/ArxTypeTraits/ArxTypeTraits/type_traits.h:153:19: note: candidate: constexpr T&& arx::stdx::forward(typename arx::stdx::remove_reference<T>::type&) [with T = const ArduinoJson::V702L1::StaticJsonDocument<200>&; typename arx::stdx::remove_reference<T>::type = const ArduinoJson::V702L1::StaticJsonDocument<200>]
     constexpr T&& forward(typename remove_reference<T>::type& t) noexcept
                   ^~~~~~~
In file included from /home/ken/Arduino/libraries/ArduinoJson/src/ArduinoJson/Memory/StringPool.hpp:10:0,
                 from /home/ken/Arduino/libraries/ArduinoJson/src/ArduinoJson/Memory/ResourceManager.hpp:8,
                 from /home/ken/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/JsonVariantConst.hpp:10,
                 from /home/ken/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:9,
                 from /home/ken/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ElementProxy.hpp:7,
                 from /home/ken/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/JsonArray.hpp:7,
                 from /home/ken/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:29,
                 from /home/ken/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /tmp/.arduinoIDE-unsaved202413-2067-80ztgm.3pekm/sketch_feb3a/sketch_feb3a.ino:1:
/home/ken/Arduino/libraries/ArduinoJson/src/ArduinoJson/Polyfills/utility.hpp:14:5: note: candidate: T&& ArduinoJson::V702L1::detail::forward(typename ArduinoJson::V702L1::detail::remove_reference<T>::type&) [with T = const ArduinoJson::V702L1::StaticJsonDocument<200>&; typename ArduinoJson::V702L1::detail::remove_reference<T>::type = const ArduinoJson::V702L1::StaticJsonDocument<200>]
 T&& forward(typename remove_reference<T>::type& t) noexcept {
     ^~~~~~~

exit status 1

Compilation error: exit status 1
```

hideakitai, 2024-02-03:
What's the version of ArduinoJson?

KennethThompson, 2024-02-03:
> What's the version of ArduinoJson?

7.0.2

KennethThompson, 2024-02-03:
> What's the version of ArduinoJson?

I tried compiling examples from ArduinoJson and I can't get the error to reappear- not sure if that is helpful.

KennethThompson, 2024-02-03:
> What's the version of ArduinoJson?

Apparently, 'Arduino Nano' board also triggers the compile errors above.  Maybe it is using the same DebugLog conflicting code as the 2560?

hideakitai, 2024-02-03:
forward() may be conflicting with ArduinoJson (and ArxTypeTraits). Could you try ArduinoJson v6.x.x?

KennethThompson, 2024-02-03:
> forward() may be conflicting with ArduinoJson (and ArxTypeTraits). Could you try ArduinoJson v6.x.x?

Funny you mention that. I tried a few different 6.x versions and all of them have the same compiler error.  For now, we're moving the #include <ArduinoJson.h> before the MsgPacketizer include as we are not using your automatic json parsing.

hideakitai, 2024-02-03:
It should compile up to ArduinoJson v6.20.1, have you tried that?

Also, why are you including ArduinoJson when you don't need it, it should only be included in examples that use ArduinoJson, so please remove it.

KennethThompson, 2024-02-04:
> It should compile up to ArduinoJson v6.20.1, have you tried that?
> 
> Also, why are you including ArduinoJson when you don't need it, it should only be included in examples that use ArduinoJson, so please remove it.

The example still does not compile with 6.20.1 installed.  At some point, we decided to not use the integrated JSON abilities of MsgPacketizer and just forgot.  We still use ArduinoJson, but were not concerned with whether we included it before or after MsgPacketizer.  It would be nice if we could use the ArduinoJson features of MsgPacketizer in the future- but maybe that will be limited to specific boards if we need it.

hideakitai, 2024-02-04:
Fixed. Please update your dependencies

- DebugLog v0.8.0
- ArxTypeTraits v0.3.1

closed 2024-02-04 by hideakitai

KennethThompson, 2024-02-04:
Excellent! Thank you- will be testing ASAP