hideakitai/MsgPacketizer #17: How can one turn on logging, such as output via the serial port of Arduino for msgpacketizer?

KennethThompson, 2024-02-03:
How can I enable logging, such as output to the serial port, so I can understand what MsgPacketizer is up to? I want to be able to determine if its receiving messages it cannot decode for whatever reason.

Thanks very much!

hideakitai, 2024-02-04:
Please declare #define MSGPACKETIZER_DEBUGLOG_ENABLE and use Serial1 for MsgPacketizer

KennethThompson, 2024-02-04:
Ok.  So I assume if we use Serial in the MsgPacketizer, the Debug automatically stops being output?  I'll try it with Serial1 as the protocol carrier

KennethThompson, 2024-02-04:
> Ok. So I assume if we use Serial in the MsgPacketizer, the Debug automatically stops being output? I'll try it with Serial1 as the protocol carrier

As an additional, un-related note, FastCRC Author has merged in my pull that brought support for a bunch of modern boards.  Hopefully that helps future users of your library use their Minimas, RP2040, R4 wifis, etc.  Thank you for your hard work on this library

hideakitai, 2024-02-12:
If you still need help, please feel free to reopen this issue.

closed 2024-02-12 by hideakitai