hideakitai/MsgPacketizer #19: V0.5.0; breaks compatibility with Grand Central M4 / SAMD21 / SAMD51

KennethThompson, 2024-02-06:
Seems that all SAMD21 and SAMD51 boards cannot build latest V0.5.0.  I only have a Adafruit Grand Central M4, but can confirm V0.5.0 breaks compatibility when I target other SAMD51/SAMD21 boards as well such as the Feather M0 (SAMD21) and Feather M4 (SAMD21).  Downgrading to V0.4.7 fixes the compilation issues.

Sketch I used to confirm:

```
// #define MSGPACKETIZER_DEBUGLOG_ENABLE
#include <MsgPacketizer.h>

const uint8_t recv_index = 0x12;
const uint8_t send_index = 0x34;

void setup() {
    Serial.begin(115200);
    delay(2000);

    // - only one interface (serial, udp, tcp, etc.) is available for manual subscription
    //   because MsgPacketizer cannot indetify which data is from which device
    // - publisher is not available for unsupported data stream (manual operation)
    MsgPacketizer::subscribe_manual(recv_index,
        [&](const MsgPack::arr_size_t& sz, const int i, const float f, const String& s) {
            if (sz.size() == 3)  // if array size is correct
            {
                // modify input and send back as array format
                static int count = 0;
                String str = s + " " + sz.size() + " " + count++;

                // encode your data manually and get binary packet from MsgPacketizer
                const auto& packet = MsgPacketizer::encode_arr(send_index, i * 2, f * 2, str);

                // send the packet data with your interface
                Serial.write(packet.data.data(), packet.data.size());
            }
        });
}

void loop() {
    // you should feed the received data manually to MsgPacketizer
    const size_t size = Serial.available();
    if (size) {
        uint8_t* data = new uint8_t[size];
        Serial.readBytes((char*)data, size);

        // feed your binary data to MsgPacketizer manually
        // if data has successfully received and decoded,
        // subscribed callback will be called
        MsgPacketizer::feed(data, size);

        delete[] data;
    }
}

```

```
In file included from /home/ken/Arduino/libraries/DebugLog/DebugLogRestoreState.h:4,
                 from /home/ken/Arduino/libraries/DebugLog/DebugLog.h:53,
                 from /home/ken/Arduino/libraries/MsgPacketizer/MsgPacketizer.h:64,
                 from /tmp/.arduinoIDE-unsaved202415-27510-1ei8fpl.3ab7/simple_bind_lambda_manual/simple_bind_lambda_manual.ino:2:
/home/ken/Arduino/libraries/DebugLog/DebugLogEnable.h:50:4: warning: #warning "Defaulting to a log level of: DEBUGLOG_DEFAULT_LOG_LEVEL_TRACE" [-Wcpp]
   50 |   #warning "Defaulting to a log level of: DEBUGLOG_DEFAULT_LOG_LEVEL_TRACE"
      |    ^~~~~~~
In file included from /home/ken/Arduino/libraries/DebugLog/DebugLogRestoreState.h:4,
                 from /home/ken/Arduino/libraries/MsgPack/MsgPack.h:19,
                 from /home/ken/Arduino/libraries/MsgPacketizer/MsgPacketizer.h:73,
                 from /tmp/.arduinoIDE-unsaved202415-27510-1ei8fpl.3ab7/simple_bind_lambda_manual/simple_bind_lambda_manual.ino:2:
/home/ken/Arduino/libraries/DebugLog/DebugLogEnable.h:50:4: warning: #warning "Defaulting to a log level of: DEBUGLOG_DEFAULT_LOG_LEVEL_TRACE" [-Wcpp]
   50 |   #warning "Defaulting to a log level of: DEBUGLOG_DEFAULT_LOG_LEVEL_TRACE"
      |    ^~~~~~~
In file included from /home/ken/Arduino/libraries/DebugLog/DebugLogRestoreState.h:4,
                 from /home/ken/Arduino/libraries/MsgPacketizer/MsgPacketizer.h:126,
[... 13 lines cut ...]
                 from /home/ken/Arduino/libraries/DebugLog/DebugLog.h:14,
                 from /home/ken/Arduino/libraries/MsgPacketizer/MsgPacketizer.h:64,
                 from /tmp/.arduinoIDE-unsaved202415-27510-1ei8fpl.3ab7/simple_bind_lambda_manual/simple_bind_lambda_manual.ino:2:
/home/ken/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/arm-none-eabi/include/c++/9.2.1/ext/new_allocator.h: In instantiation of 'void __gnu_cxx::new_allocator<_Tp>::construct(_Up*, _Args&& ...) [with _Up = std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> >; _Args = {arx::stdx::pair<unsigned char, std::function<void(const unsigned char*, unsigned int)> >}; _Tp = std::_Rb_tree_node<std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> > >]':
/home/ken/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/arm-none-eabi/include/c++/9.2.1/bits/alloc_traits.h:482:2:   required from 'static void std::allocator_traits<std::allocator<_CharT> >::construct(std::allocator_traits<std::allocator<_CharT> >::allocator_type&, _Up*, _Args&& ...) [with _Up = std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> >; _Args = {arx::stdx::pair<unsigned char, std::function<void(const unsigned char*, unsigned int)> >}; _Tp = std::_Rb_tree_node<std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> > >; std::allocator_traits<std::allocator<_CharT> >::allocator_type = std::allocator<std::_Rb_tree_node<std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> > > >]'
/home/ken/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/arm-none-eabi/include/c++/9.2.1/bits/stl_tree.h:614:32:   required from 'void std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_construct_node(std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_Link_type, _Args&& ...) [with _Args = {arx::stdx::pair<unsigned char, std::function<void(const unsigned char*, unsigned int)> >}; _Key = unsigned char; _Val = std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> >; _KeyOfValue = std::_Select1st<std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> > >; _Compare = std::less<unsigned char>; _Alloc = std::allocator<std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> > >; std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_Link_type = std::_Rb_tree_node<std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> > >*]'
/home/ken/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/arm-none-eabi/include/c++/9.2.1/bits/stl_tree.h:631:4:   required from 'std::_Rb_tree_node<_Val>* std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_create_node(_Args&& ...) [with _Args = {arx::stdx::pair<unsigned char, std::function<void(const unsigned char*, unsigned int)> >}; _Key = unsigned char; _Val = std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> >; _KeyOfValue = std::_Select1st<std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> > >; _Compare = std::less<unsigned char>; _Alloc = std::allocator<std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> > >; std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_Link_type = std::_Rb_tree_node<std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> > >*]'
/home/ken/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/arm-none-eabi/include/c++/9.2.1/bits/stl_tree.h:2408:13:   required from 'std::pair<std::_Rb_tree_iterator<_Val>, bool> std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_emplace_unique(_Args&& ...) [with _Args = {arx::stdx::pair<unsigned char, std::function<void(const unsigned char*, unsigned int)> >}; _Key = unsigned char; _Val = std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> >; _KeyOfValue = std::_Select1st<std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> > >; _Compare = std::less<unsigned char>; _Alloc = std::allocator<std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> > >]'
/home/ken/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/arm-none-eabi/include/c++/9.2.1/bits/stl_map.h:575:64:   required from 'std::pair<typename std::_Rb_tree<_Key, std::pair<const _Key, _Tp>, std::_Select1st<std::pair<const _Key, _Tp> >, _Compare, typename __gnu_cxx::__alloc_traits<_Allocator>::rebind<std::pair<const _Key, _Tp> >::other>::iterator, bool> std::map<_Key, _Tp, _Compare, _Alloc>::emplace(_Args&& ...) [with _Args = {arx::stdx::pair<unsigned char, std::function<void(const unsigned char*, unsigned int)> >}; _Key = unsigned char; _Tp = std::function<void(const unsigned char*, unsigned int)>; _Compare = std::less<unsigned char>; _Alloc = std::allocator<std::pair<const unsigned  [... cut ...]
/home/ken/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/arm-none-eabi/include/c++/9.2.1/ext/new_allocator.h:145:20: error: no matching function for call to 'std::pair<const unsigned char, std::function<void(const unsigned char*, unsigned int)> >::pair(arx::stdx::pair<unsigned char, std::function<void(const unsigned char*, unsigned int)> >)'
  145 |  noexcept(noexcept(::new((void *)__p)
      |                    ^~~~~~~~~~~~~~~~~~
  146 |        _Up(std::forward<_Args>(__args)...)))
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
<<< TRUNCATED FOR BREVITY >>>

exit status 1

Compilation error: exit status 1
```

KennethThompson, 2024-02-07:
I can further confirm that the build errors also occur when compiling for the Arduino R4 Minima

hideakitai, 2024-02-12:
The dependecies are not correctly installed. Please try v0.5.1 with the latest release of dependencies (clone from repo because they are not released to Arduino IDE yet)

closed 2024-02-12 by hideakitai

AndBondStyle, 2024-03-31:
Still not building for me on Teensy 4.0, tried both 0.5.0 and 0.5.1. Version 0.4.7 somehow got removed (intentionally?) from platformio registry, so I had to resort to `pio pkg install https://github.com/hideakitai/MsgPacketizer#0.4.7`

<details>
  <summary>Logs</summary>

```
Processing main (platform: teensy; board: teensy40; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing hideakitai/MsgPacketizer @ ^0.5.1
Unpacking  [####################################]  100%
Library Manager: MsgPacketizer@0.5.1 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing hideakitai/DebugLog @ *
Unpacking  [####################################]  100%
Library Manager: DebugLog@0.8.3 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing hideakitai/ArxContainer @ >=0.6.0
Unpacking  [####################################]  100%
Library Manager: ArxContainer@0.6.0 has been installed!
Library Manager: Installing hideakitai/ArxTypeTraits @ *
Unpacking  [####################################]  100%
Library Manager: ArxTypeTraits@0.3.1 has been installed!
Library Manager: Installing hideakitai/MsgPack @ *
Unpacking  [####################################]  100%
Library Manager: MsgPack@0.4.2 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing hideakitai/Packetizer @ *
Unpacking  [####################################]  100%
Library Manager: Packetizer@0.4.5 has been installed!
Library Manager: Installing pierremolinaro/ACAN_T4 @ ^1.1.8
Unpacking  [####################################]  100%
Library Manager: ACAN_T4@1.1.8 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
PLATFORM: Teensy (5.0.0) > Teensy 4.0
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 1.94MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.159.0 (1.59) 
 - tool-teensy @ 1.159.0 (1.59) 
 - toolchain-gccarmnoneeabi-teensy @ 1.110301.0 (11.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 99 compatible libraries
Scanning dependencies...
Dependency Graph
|-- MsgPacketizer @ 0.5.1
|-- ACAN_T4 @ 1.1.8
|-- Servo @ 1.1.2
Building in release mode

...

In file included from .pio/libdeps/main/MsgPacketizer/MsgPacketizer.h:115,
                 from src/main.cpp:3:
.pio/libdeps/main/MsgPacketizer/MsgPacketizer/Publisher.h: In member function 'void arduino::msgpack::msgpacketizer::PackerManager::send(const arduino::msgpack::msgpacketizer::Destination&, arduino::msgpack::msgpacketizer::PublishElementRef)':
.pio/libdeps/main/MsgPacketizer/MsgPacketizer/Publisher.h:227:41: error: no matching function for call to 'send(UDP&, const str_t&, const uint16_t&, const uint8_t&, const uint8_t*, size_t)'
  227 |                         Packetizer::send(
      |                         ~~~~~~~~~~~~~~~~^
  228 |                             *reinterpret_cast<UDP*>(dest.stream),
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  229 |                             dest.ip,
      |                             ~~~~~~~~     
  230 |                             dest.port,
      |                             ~~~~~~~~~~   
  231 |                             dest.index,
      |                             ~~~~~~~~~~~  
  232 |                             encoder.data(),
      |                             ~~~~~~~~~~~~~~~
  233 |                             encoder.size());
[... 161 lines cut ...]
~~~~~~~~~~~~~~~~~~~
In file included from .pio/libdeps/main/MsgPacketizer/MsgPacketizer.h:72,
                 from src/main.cpp:3:
.pio/libdeps/main/Packetizer/Packetizer.h:216:17: note: candidate: 'template<class Encoding> void ht::serial::packetizer::send(ht::serial::packetizer::StreamType&, uint8_t, const uint8_t*, size_t, bool)'
  216 |     inline void send(StreamType& stream, const uint8_t index, const uint8_t* data, const size_t size, const bool b_crc = encoding::default_option.b_crc)
      |                 ^~~~
.pio/libdeps/main/Packetizer/Packetizer.h:216:17: note:   template argument deduction/substitution failed:
In file included from .pio/libdeps/main/MsgPacketizer/MsgPacketizer.h:115,
                 from src/main.cpp:3:
.pio/libdeps/main/MsgPacketizer/MsgPacketizer/Publisher.h:571:33: note:   candidate expects 5 arguments, 6 provided
  571 |                 Packetizer::send(stream, ip, port, index, packer.data(), packer.size());
      |                 ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from .pio/libdeps/main/MsgPacketizer/MsgPacketizer.h:72,
                 from src/main.cpp:3:
.pio/libdeps/main/Packetizer/Packetizer.h:223:17: note: candidate: 'template<class Encoding> void ht::serial::packetizer::send(ht::serial::packetizer::StreamType&, const uint8_t*, size_t, bool)'
  223 |     inline void send(StreamType& stream, const uint8_t* data, const size_t size, const bool b_crc = encoding::default_option.b_crc)
      |                 ^~~~
.pio/libdeps/main/Packetizer/Packetizer.h:223:17: note:   template argument deduction/substitution failed:
In file included from .pio/libdeps/main/MsgPacketizer/MsgPacketizer.h:115,
                 from src/main.cpp:3:
.pio/libdeps/main/MsgPacketizer/MsgPacketizer/Publisher.h:571:33: note:   candidate expects 4 arguments, 6 provided
  571 |                 Packetizer::send(stream, ip, port, index, packer.data(), packer.size());
      |                 ~~~~~Compiling .pio/build/main/FrameworkArduino/serialEvent7.cpp.o
 ```
</details>

hideakitai, 2024-04-04:
Which example do you use with platformio?

AndBondStyle, 2024-04-05:
Minimal non-working example: https://gist.github.com/AndBondStyle/fc6703e017b5125dd4db2663a78e9983

Also tried [nested_custom.ino](https://github.com/hideakitai/MsgPacketizer/blob/main/examples/nested_custom/arduino/nested_custom/nested_custom.ino) example, failed in the same way

hideakitai, 2024-04-28:
0.5.1 should be compiled if the dependencies are correctly installed. 

https://github.com/hideakitai/MsgPacketizer/actions/runs/7902011313

Please confirm the installation of dependencies.