hideakitai/MsgPacketizer #24: How to not use the Network

MonoLogueChi, 2024-07-22:
```cpp
#if defined(ARDUINO) || defined(OF_VERSION_MAJOR) || defined(SERIAL_H)
#define MSGPACKETIZER_ENABLE_STREAM
#ifdef ARDUINO  // TODO: support more platforms
#if defined(ESP_PLATFORM) || defined(ESP8266) || defined(ARDUINO_AVR_UNO_WIFI_REV2)                             \
    || defined(ARDUINO_SAMD_MKRWIFI1010) || defined(ARDUINO_SAMD_MKRVIDOR4000) || defined(ARDUINO_SAMD_MKR1000) \
    || defined(ARDUINO_SAMD_NANO_33_IOT)
#define MSGPACKETIZER_ENABLE_WIFI
#endif
#if defined(ESP_PLATFORM) || defined(ESP8266) || !defined(MSGPACKETIZER_ENABLE_WIFI)
#define MSGPACKETIZER_ENABLE_ETHER
#endif
#endif
#endif

#if defined(MSGPACKETIZER_ENABLE_ETHER) || defined(MSGPACKETIZER_ENABLE_WIFI)
#define MSGPACKETIZER_ENABLE_NETWORK
#include <Udp.h>
#include <Client.h>
#endif
```

If I read it correctly, `MSGPACKETIZER_ENABLE_ETHER` is defined when `MSGPACKETIZER_ENABLE_WIFI` is not defined.

Does this mean that `MSGPACKETIZER_ENABLE_NETWORK` must be defined? But `#include <Udp.h>` will report an error on some development boards, such as ch32v307

hideakitai, 2024-07-23:
Either `MSGPACKETIZER_ENABLE_ETHER` or `MSGPACKETIZER_ENABLE_WIFI` will be defined for all boards. So, `MSGPACKETIZER_ENABLE_NETWORK` will be defined for all boards.

The Arduino framework usually has `Udp.h` and `Client.h`, so almost all boards are available for this library. It seems that arduino_core_ch32 also has them.

https://github.com/openwch/arduino_core_ch32/tree/main/cores/arduino

hideakitai, 2024-07-23:
Please give me the error details

MonoLogueChi, 2024-07-23:
error:

```
In file included from D:\Users\mc\Documents\Arduino\Projects\msgpack-demo\msgpack-demo.ino:3:
d:\Users\mc\Documents\Arduino\libraries\MsgPacketizer/MsgPacketizer.h:22:10: fatal error: Udp.h: No such file or directory
 #include <Udp.h>
          ^~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1
```

ino:

```ino
#include <Arduino.h>
#include <MsgPack.h>
#include <MsgPacketizer.h>
// #include <U8g2lib.h>

// U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R2, /* reset=*/U8X8_PIN_NONE, /* clock=*/9, /* data=*/8);
// U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R2, /* reset=*/U8X8_PIN_NONE, /* clock=*/PB10, /* data=*/PB11);

struct MessgaeData {
  // MsgPack::arr_t<uint8_t> Id;
  uint8_t Type;
  uint8_t Code;
  MsgPack::arr_t<uint8_t> Data;
  MSGPACK_DEFINE(Type, Code, Data);
};
MessgaeData data = {
  .Type = 0,
  .Code = 0,
  .Data = { 1, 2, 3, 4 }
};
byte send_index = 0x01;
byte recv_index = 0x01;

void setup() {
  Serial.begin(115200);
  // u8g2.setBusClock(800000);
  // u8g2.begin();

  // MsgPacketizer::publish(Serial, send_index, data)
  //   ->setFrameRate(1.f);

  MsgPacketizer::subscribe(Serial, recv_index,
                           [&](const MessgaeData& md) {
                            //  u8g2.drawXBMP(0, 0, 88, 64, md.Data.data());
                            //  u8g2.updateDisplay();
                           });
}

void loop() {
  MsgPacketizer::update();
}
```

When I comment out 

```cpp
// #define PACKETIZER_ENABLE_NETWORK
// #include <Udp.h>
// #include <Client.h>
```

```cpp
// #define MSGPACKETIZER_ENABLE_NETWORK
// #include <Udp.h>
// #include <Client.h>
```

 in `Packetizer.h` and `MsgPacketizer.h`, it compiles fine

hideakitai, 2024-07-23:
How about examples?

MonoLogueChi, 2024-07-23:
> How about examples?

this https://github.com/hideakitai/MsgPacketizer/issues/24#issuecomment-2244923510

And there is a simpler

```ino
#include <Arduino.h>
#include <MsgPacketizer.h>

void setup() {
}

void loop() {
}
```

[image]

hideakitai, 2024-07-24:
Simply, `Ethernet` library and related components are not included (not a compile target) in the arduino-core-ch32. It's a problem of ch32 core

hideakitai, 2024-07-24:
Now you can use the following code with [Packtizer v0.8.3](https://github.com/hideakitai/Packetizer/releases/tag/v0.8.3) and [MsgPacketizer v0.5.3](https://github.com/hideakitai/MsgPacketizer/releases/tag/v0.5.3)

```cpp
#define MSGPACKETIZER_DISABLE_NETWORK
#include <MsgPacketizer.h>
```

closed 2024-07-25 by MonoLogueChi